JeffyCN/mirrors #3: U-Boot fails to generate for PX30

StefanLellwitz, 2022-01-06:
Not possible to build U-Boot for PX30.
<pre>
| Platform PX30 is build OK, with exist .config
| 
| Thu Jan  6 10:27:56 AM UTC 2022
| NOTE: u-boot: Generating idblock.img from loader.bin
| unpack entry(px30_ddr_333MHz_v1)
| unpack entry(px30_usbplug_v1)
| unpack entry(FlashData)
| unpack entry(FlashBoot)
| unpack success
<b>| Error: SPL image is too large (size 0x3000 than 0x2800)</b>
<b>| Error: Bad parameters for image type</b>
| Usage: ./tools/mkimage -l image
|           -l ==> list image header information
|        ./tools/mkimage [-x] -A arch -O os -T type -C comp -a addr -e ep -n name -d data_file[:data_file...] image
|           -A ==> set architecture to 'arch'
|           -O ==> set operating system to 'os'
|           -T ==> set image type to 'type'
|           -C ==> set compression type 'comp'
|           -a ==> set load address to 'addr' (hex)
|           -e ==> set entry point to 'ep' (hex)
|           -n ==> set image name to 'name'
|           -d ==> use image data from 'datafile'
|           -x ==> set XIP (execute in place)
|        ./tools/mkimage [-D dtc_options] [-f fit-image.its|-f auto|-F] [-b <dtb> [-b <dtb>]] [-i <ramdisk.cpio.gz>] fit-image
|            <dtb> file is used with -f auto, it may occur multiple times.
|           -D => set all options for device tree compiler
|           -f => input filename for FIT source
|           -i => input filename for ramdisk file
|           -v => set FIT image version in decimal
| Signing / verified boot not supported (CONFIG_FIT_SIGNATURE undefined)
|        ./tools/mkimage -V ==> print version information and exit
| Use '-T list' to see a list of available image types
| WARNING: exit code 1 from a shell command.
</pre>
This problem exists since [413d2f5](https://github.com/JeffyCN/mirrors/commit/413d2f55e56ba99d41e2cda1d54c27bd6603a3bf).  
Before `px30_ddr_333MHz_v1.15.bin` was exactly 10.240 Bytes (= 0x2800) and everything was fine.
But since update to `px30_ddr_333MHz_v1.16.bin` it is 10.24**8** Bytes and therefore does not fit.

JeffyCN, 2022-01-06:
please contact the maintainer about this issue, likely chenjh@rock-chips.com.

this repo is just a mirror, i don't know much about this....as far as i know, the size limit should be the chip's sram size, and bootroom would load ddr init code(spl) into it before ddr inited.

so guessing they need to double check the sram size or truncate the ddr bin.

there seems a few meaningless padding at the end:
00027f0 0000 0000 0030 0000 0004 0000 ffff ffff
0002800 ffff ffff beaf dead

closed 2022-01-06 by JeffyCN

JeffyCN, 2022-01-07:
hmm, tested on rk3399, the ddr binary's padding bytes are important.

checking px30's TRM, the "Chapter 6 Embedded SRAM" says "SYSTEM_SRAM Provide 16KB access space", so guessing the 0x2800 should be 0x4000

StefanLellwitz, 2022-01-07:
Hi Jeffy, thanks for reviewing this.

> checking px30's TRM, the "Chapter 6 Embedded SRAM" says "SYSTEM_SRAM Provide 16KB access space", so guessing the 0x2800 should be 0x4000

This was the whole reason for my Pull Request https://github.com/JeffyCN/mirrors/pull/4. I found this information of 16K SYSTEM_SRAM in PX30 datasheet. But I wasn't sure about that. So the absolute minimum which works is 12K (0x3000).  
But should be rather 0x4000 based on this information I guess...
Anyway, [this](https://github.com/JeffyCN/mirrors/commit/cab28f403dae8ca7690a93e8ff7433749b79e91f#diff-a574ea9d0ea21f8e37bbe06be8c06fc23e434824ceec520f39397579f7de727d) seems to be wrong.  

I will contact U-Boot maintainer as you suggested.

StefanLellwitz, 2022-01-07:
> 0x4000 <b>- (bootrom's reserved size(mostly stack, and likely to be 4K))</b>

Sorry, didn't get it. Would be 0x3000 then.

JeffyCN, 2022-01-07:
> > 0x4000 - (bootrom's reserved size(mostly stack, and likely to be 4K))
> 
> Sorry, didn't get it. Would be 0x3000 then.

right, the bootrom is close-source,  so the reserved size(6k) in the original commit might just be a guess value.

anyway, the bootrom guys did confirm it's 4k reserved in px30/3326. the fix patch is going through the rockchip internal review flow now.