JeffyCN/mirrors #12: gstreamer-rockchip cannot decode NV16

avafinger, 2023-06-24:
@JeffyCN 

I updated gstreamer-rockchip and it cannot decode NV16 anymore.
My previous version that worked was with the commit: https://github.com/JeffyCN/mirrors/commit/2ed1e68b0aa77728b1d493344d8e62a04b1b64e0

> gst-launch-1.0 filesrc location=/apps/Videos/Poltergeist_DOM_TrailerD-1080p-HDTN.mp4 ! qtdemux ! h264parse ! mppvideodec arm-afbc=1 ! waylandsink

JeffyCN, 2023-06-25:
the waylandsink requires dma-feature for this kind of usage:

https://github.com/JeffyCN/mirrors/commit/4d2ae6b3d8c4fdbc8f2971ce7b64ecaf0a097412

avafinger, 2023-06-25:
I've cloned the gstreamer-rockchip which has all commits.

This was also working:

> gst-launch-1.0 filesrc location=/apps/Videos/Poltergeist_DOM_TrailerD-1080p-HDTN.mp4 ! qtdemux ! h264parse ! mppvideodec ! glimagesink

JeffyCN, 2023-06-25:
export GST_MPP_DEC_DMA_FEATURE=1

or set dma-feature=true for mppvideodec

closed 2023-06-25 by avafinger