JeffyCN/mirrors #16: Out of memory

ParhamJ, 2023-12-13:
Hello. I upgraded the Rockchip GStreamer library in NanoPi R6C Debian 11 (5.10.160), but it was when I ran my transcoding app. I have a question, Is there any requirement for updating the lib (libgstrockchipmpp.so)? And how can I resolve the issue? 

Also, my app was developed based on the previous version (2021), and it works based on that lib, but because of some errors that I saw in the system log I wanted to upgrade it to the latest version:

This is the issue after updating the new lib (libgstrockchipmpp):
rga_api: 1.9.1
rga_driver: 1.2.26
```
Dec 13 16:29:56 NanoPi-R6C kernel: mali fb000000.gpu: OOM notifier: dev mali0  35564 kB
Dec 13 16:29:56 NanoPi-R6C kernel: mali fb000000.gpu: OOM notifier: tsk Xorg  tgid (771)  pid (771) 35532 kB
Dec 13 16:29:56 NanoPi-R6C kernel: stamp thread invoked oom-killer: gfp_mask=0x1100cca(GFP_HIGHUSER_MOVABLE), order=0, oom_score_adj=0
Dec 13 16:29:56 NanoPi-R6C kernel: CPU: 5 PID: 2072 Comm: stamp thread Tainted: G           O      5.10.160 #194
Dec 13 16:29:56 NanoPi-R6C kernel: Hardware name: FriendlyElec NanoPi R6C (DT)
Dec 13 16:29:56 NanoPi-R6C kernel: Call trace:
Dec 13 16:29:56 NanoPi-R6C kernel:  dump_backtrace+0x0/0x1d0
Dec 13 16:29:56 NanoPi-R6C kernel:  show_stack+0x1c/0x24
Dec 13 16:29:56 NanoPi-R6C kernel:  dump_stack_lvl+0xc8/0xec
Dec 13 16:29:56 NanoPi-R6C kernel:  dump_stack+0x14/0x50
Dec 13 16:29:56 NanoPi-R6C kernel:  dump_header+0x4c/0x1ec
Dec 13 16:29:56 NanoPi-R6C kernel:  oom_kill_process+0x1c0/0x1c4
Dec 13 16:29:56 NanoPi-R6C kernel:  out_of_memory+0x1e0/0x414
Dec 13 16:29:56 NanoPi-R6C kernel:  __alloc_pages_may_oom+0x114/0x190
Dec 13 16:29:56 NanoPi-R6C kernel:  __alloc_pages_nodemask+0x984/0xcac
Dec 13 16:29:56 NanoPi-R6C kernel:  pagecache_get_page+0x1b4/0x384
Dec 13 16:29:56 NanoPi-R6C kernel:  filemap_fault+0x6ec/0x940
Dec 13 16:29:56 NanoPi-R6C kernel:  ext4_filemap_fault+0x38/0x900
Dec 13 16:29:56 NanoPi-R6C kernel:  __do_fault+0x40/0x160
Dec 13 16:29:56 NanoPi-R6C kernel:  handle_pte_fault+0x8b4/0xdc0
Dec 13 16:29:56 NanoPi-R6C kernel:  __handle_mm_fault+0x10c/0x2d0
Dec 13 16:29:56 NanoPi-R6C kernel:  handle_mm_fault+0xc4/0x234
Dec 13 16:29:56 NanoPi-R6C kernel:  do_page_fault+0x2b8/0x4c0
Dec 13 16:29:56 NanoPi-R6C kernel:  do_translation_fault+0xac/0xc0
Dec 13 16:29:56 NanoPi-R6C kernel:  do_mem_abort+0x44/0xb0
Dec 13 16:29:56 NanoPi-R6C kernel:  el0_ia+0x68/0xac
Dec 13 16:29:56 NanoPi-R6C kernel:  el0_sync_handler+0x11c/0x120
Dec 13 16:29:56 NanoPi-R6C kernel:  el0_sync+0x15c/0x180
Dec 13 16:29:56 NanoPi-R6C kernel: Mem-Info:
Dec 13 16:29:56 NanoPi-R6C kernel: active_anon:221 inactive_anon:873839 isolated_anon:0
                                    active_file:111 inactive_file:57 isolated_file:0
                                    unevictable:805 dirty:0 writeback:0
                                    slab_reclaimable:10933 slab_unreclaimable:12498
                                    mapped:273 shmem:1520 pagetables:3774 bounce:0
                                    free:19107 free_pcp:419 free_cma:0
Dec 13 16:29:56 NanoPi-R6C kernel: Node 0 active_anon:884kB inactive_anon:3495356kB active_file:444kB inactive_file:228kB unevictable:3220kB isolated(anon):0kB isolated(file):0kB mapped:1092kB dirty:0kB writeback:0kB shmem:6080kB shmem_thp: 0kB shmem_pmdmapped: 0kB anon_thp: 0kB writeback_tmp:0kB kernel_stack:7088kB all_unreclaimable? no
Dec 13 16:29:56 NanoPi-R6C kernel: DMA free:70852kB min:70396kB low:87992kB high:105588kB reserved_highatomic:0KB active_anon:264kB inactive_anon:3406488kB active_file:100kB inactive_file:0kB unevictable:3200kB writepending:0kB present:3913728kB managed:3815692kB mlocked:64kB pagetables:13580kB bounce:0kB free_pcp:1412kB local_pcp:248kB free_cma:0kB
Dec 13 16:29:56 NanoPi-R6C kernel: lowmem_reserve[]: 0 0 175 175
Dec 13 16:29:56 NanoPi-R6C kernel: Normal free:5576kB min:3328kB low:4160kB high:4992kB reserved_highatomic:0KB active_anon:620kB inactive_anon:88868kB active_file:252kB inactive_file:104kB unevictable:20kB writepending:0kB present:262144kB managed:179836kB mlocked:0kB pagetables:1516kB bounce:0kB free_pcp:264kB local_pcp:76kB free_cma:0kB
Dec 13 16:29:56 NanoPi-R6C kernel: lowmem_reserve[]: 0 0 0 0
Dec 13 16:29:56 NanoPi-R6C kernel: DMA: 394*4kB (UME) 299*8kB (UME) 271*16kB (UE) 248*32kB (UME) 129*64kB (UE) 63*128kB (UE) 39*256kB (UE) 25*512kB (UME) 10*1024kB (UME) 2*2048kB (U) 0*4096kB = 69680kB
Dec 13 16:29:56 NanoPi-R6C kernel: Normal: 363*4kB (U) 304*8kB (UE) 61*16kB (UE) 20*32kB (U) 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 5500kB
Dec 13 16:29:56 NanoPi-R6C kernel: Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=1048576kB
Dec 13 16:29:56 NanoPi-R6C kernel: Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=32768kB
Dec 13 16:29:56 NanoPi-R6C kernel: Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=2048kB
Dec 13 16:29:56 NanoPi-R6C kernel: Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=64kB
Dec 13 16:29:56 NanoPi-R6C kernel: 1682 total pagecache pages
Dec 13 16:29:56 NanoPi-R6C kernel: 0 pages in swap cache
Dec 13 16:29:56 NanoPi-R6C kernel: Swap cache stats: add 0, delete 0, find 0/0
Dec 13 16:29:56 NanoPi-R6C kernel: Free swap  = 0kB
Dec 13 16:29:56 NanoPi-R6C kernel: Total swap = 0kB
Dec 13 16:29:56 NanoPi-R6C kernel: 1043968 pages RAM
Dec 13 16:29:56 NanoPi-R6C kernel: 0 pages HighMem/MovableOnly
Dec 13 16:29:56 NanoPi-R6C kernel: 45086 pages reserved
Dec 13 16:29:56 NanoPi-R6C kernel: 2048 pages cma reserved
Dec 13 16:29:56 NanoPi-R6C kernel: Tasks state (memory values in pages):
Dec 13 16:29:56 NanoPi-R6C kernel: [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
Dec 13 16:29:56 NanoPi-R6C kernel: [    225]     0   225      206        1    28672        0             0 bpfilter_umh
Dec 13 16:29:56 NanoPi-R6C kernel: [    377]     0   377    16376      281    98304        0          -250 systemd-journal
Dec 13 16:29:56 NanoPi-R6C kernel: [    421]     0   421     5077      390    57344        0         -1000 systemd-udevd
Dec 13 16:29:56 NanoPi-R6C kernel: [    582]     0   582      495       15    36864        0             0 acpid
Dec 13 16:29:56 NanoPi-R6C kernel: [    587]   104   587     2161      304    49152        0          -900 dbus-daemon
[... 55 lines cut ...]
Dec 13 16:29:56 NanoPi-R6C kernel: [   1365]  1000  1365     1461      135    45056        0             0 bash
Dec 13 16:29:56 NanoPi-R6C kernel: [   1374]  1000  1374     3991      301    61440        0             0 sshd
Dec 13 16:29:56 NanoPi-R6C kernel: [   1375]  1000  1375     1340      114    40960        0             0 sftp-server
Dec 13 16:29:56 NanoPi-R6C kernel: [   1385]     0  1385     3991      299    61440        0             0 sshd
Dec 13 16:29:56 NanoPi-R6C kernel: [   1387]     0  1387     3991      299    61440        0             0 sshd
Dec 13 16:29:56 NanoPi-R6C kernel: [   1392]  1000  1392     4076      334    65536        0             0 sshd
Dec 13 16:29:56 NanoPi-R6C kernel: [   1396]  1000  1396     1461      136    40960        0             0 bash
Dec 13 16:29:56 NanoPi-R6C kernel: [   1405]  1000  1405     3991      300    61440        0             0 sshd
Dec 13 16:29:56 NanoPi-R6C kernel: [   1406]  1000  1406     1340      114    40960        0             0 sftp-server
Dec 13 16:29:56 NanoPi-R6C kernel: [   1776]   999  1776    22002      185    65536        0             0 systemd-timesyn
Dec 13 16:29:56 NanoPi-R6C kernel: [   1973]  1000  1973    20336      173   151552        0             0 journalctl
Dec 13 16:29:56 NanoPi-R6C kernel: [   2067]  1000  2067     2752      139    57344        0             0 sudo
Dec 13 16:29:56 NanoPi-R6C kernel: [   2068]     0  2068   188593      300   147456        0             0 MediaServer
Dec 13 16:29:56 NanoPi-R6C kernel: [   2223]  1000  2223     1985      103    49152        0             0 dbus-launch
Dec 13 16:29:56 NanoPi-R6C kernel: [   2224]  1000  2224     1920       93    45056        0             0 dbus-daemon
Dec 13 16:29:56 NanoPi-R6C kernel: [   3192]     0  3192      581       15    32768        0             0 anacron
Dec 13 16:29:56 NanoPi-R6C kernel: [   3203]  1000  3203   108180     2936   167936        0             0 tracker-store
Dec 13 16:29:56 NanoPi-R6C kernel: [   3292]  1000  3292     1360       69    40960        0             0 bash
[... cut ...]
Dec 13 16:29:56 NanoPi-R6C kernel: oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/user.slice/user-1000.slice/session-3.scope,task=demo,pid=3297,uid=1000
Dec 13 16:29:56 NanoPi-R6C kernel: Out of memory: Killed process 3297 (demo) total-vm:5431168kB, anon-rss:3271208kB, file-rss:0kB, shmem-rss:4kB, UID:1000 pgtables:6936kB oom_score_adj:0
Dec 13 16:29:56 NanoPi-R6C kernel: oom_reaper: reaped process 3297 (demo), now anon-rss:0kB, file-rss:0kB, shmem-rss:4kB
Dec 13 16:30:20 NanoPi-R6C org.freedesktop.Tracker1[3203]: OK
Out of memory: Killed processDec 13 16:34:35 NanoPi-R6C anacron[3192]: Job `cron.daily' started
```

JeffyCN, 2023-12-13:
maybe try to upgrade all related packages, or maybe try to git bisect and locate the related commit

ParhamJ, 2023-12-13:
Thank you for your reply. Which packages should I upgrade to? Do I need to upgrade librockchip_mpp?

JeffyCN, 2023-12-13:
something like rockchipmpp/gst/kernel and sink related packages(xserver for xvimagesink, mali for glimagesink)

JeffyCN, 2023-12-13:
or maybe you can git bisect to a specific commit

ParhamJ, 2023-12-13:
Sure, I will check them. Thank you

bluesky2319, 2023-12-21:
I checked it with the new update, and I have the same issue:

[image]
[image]
My application scenario is based on transcoding. As I mentioned in the previous version of libgstrockchipmpp (2021) I don't have this issue.

Also, I built the project and checked in two kernel versions, 5.10 and 6.1, in Debian 11; the result is the same.
My system has already installed GStreamer version 1.18.5.
I used these commands for building the project:

```
meson setup builddir
meson compile -C builddir
```
[image]

Is there anything wrong with my build? In addition, after building the project and checking the lib, I saw the new parameter has an issue.

[image]

JeffyCN, 2023-12-29:
hmmm, that log just like the issue i fixed right now, please try to update

bluesky2319, 2023-12-30:
> hmmm, that log just like the issue i fixed right now, please try to update

The issue has been fixed. Thank you for fixing it.

Also, how can I fix the "qp-min-i" and "qp-max-i" failed message? 
[image]

closed 2023-12-31 by bluesky2319