JeffyCN/mirrors #17: cannot compile successfully

ywangwxd, 2023-12-29:
I am compiling this lib on the rk3588 box directly, not cross compile. But it always complain error messages like this. 
A lot of symbols not defined. I do not know if there is some config error. I am using ubuntu aarch64.

```

/data/librga_dev/mirrors-linux-rga-multi/core/RockchipRga.cpp: In constructor 'RockchipRga::RockchipRga()':
/data/librga_dev/mirrors-linux-rga-multi/core/RockchipRga.cpp:109:9: error: 'ALOGE' was not declared in this scope
  109 |         ALOGE("%s", RGA_API_FULL_VERSION);
      |         ^~~~~
/data/librga_dev/mirrors-linux-rga-multi/core/RockchipRga.cpp: In destructor 'RockchipRga::~RockchipRga()':
/data/librga_dev/mirrors-linux-rga-multi/core/RockchipRga.cpp:113:9: error: 'RgaDeInit' was not declared in this scope; did you mean 'RkRgaDeInit'?
  113 |         RgaDeInit(&mContext);
      |         ^~~~~~~~~
      |         RkRgaDeInit
/data/librga_dev/mirrors-linux-rga-multi/core/RockchipRga.cpp: In member function 'int RockchipRga::RkRgaInit()':
/data/librga_dev/mirrors-linux-rga-multi/core/RockchipRga.cpp:122:15: error: 'RgaInit' was not declared in this scope; did you mean 'RkRgaInit'?
  122 |         ret = RgaInit(&mContext);
      |               ^~~~~~~
      |               RkRgaInit
/data/librga_dev/mirrors-linux-rga-multi/core/RockchipRga.cpp: In member function 'void RockchipRga::RkRgaDeInit()':
/data/librga_dev/mirrors-linux-rga-multi/core/RockchipRga.cpp:133:13: error: 'RgaDeInit' was not declared in this scope; did you mean 'RkRgaDeInit'?
  133 |             RgaDeInit(&mContext);

```

ywangwxd, 2023-12-29:
It should rely on any other library right? I have removed the orignal rga include dir on the system to avoid conflict, but this did not solve my problem.

JeffyCN, 2023-12-29:
sorry, i don't know much about rga, maybe try to contact the rga maintainer(or committor).

that log is a little like including wrong headers, maybe try to use strace or other tool to locate the included headers and check them?

closed 2024-02-04 by JeffyCN