JeffyCN/mirrors #18: Two issues about the upcoming 6.1 kernel release

nyanmisaka, 2024-01-20:
I'm not sure if these have been fixed in the 6.1 final release. Therefore I report them here so that you can forward it to the corresponding developers. cc @JeffyCN 

---

1. `rk_system_heap.c` is still not ported to 6.1 yet, which causes MPP runtime to crash in DMA_HEAP allocator.

https://github.com/JeffyCN/mirrors/blob/kernel-6.1-2024_01_02/drivers/dma-buf/heaps/Makefile#L2

```shell
ls -l /dev/dma_heap/
total 0
crw------- 1 root root 251, 1 Jan 18 18:50 cma
crw------- 1 root root 251, 0 Jan 18 18:50 system
```

---

2. `rockchip_drm_gem.c` cacheable memory flag is broken during manual merging. It causes very slow copying from the decoder frame pool.

https://github.com/JeffyCN/mirrors/blob/kernel-6.1-2024_01_02/drivers/gpu/drm/rockchip/rockchip_drm_gem.c#L550

A simple fix is this:
```diff
diff --git a/drivers/gpu/drm/rockchip/rockchip_drm_gem.c b/drivers/gpu/drm/rockchip/rockchip_drm_gem.c
index 3a32a03e91f8..f386f44f825b 100644
--- a/drivers/gpu/drm/rockchip/rockchip_drm_gem.c
+++ b/drivers/gpu/drm/rockchip/rockchip_drm_gem.c
@@ -546,10 +546,6 @@ static int rockchip_drm_gem_object_mmap(struct drm_gem_object *obj,
 	int ret;
 	struct rockchip_gem_object *rk_obj = to_rockchip_obj(obj);
 
-	/* default is wc. */
-	if (rk_obj->flags & ROCKCHIP_BO_CACHABLE)
-		vma->vm_page_prot = vm_get_page_prot(vma->vm_flags);
-
 	/*
 	 * Set vm_pgoff (used as a fake buffer offset by DRM) to 0 and map the
 	 * whole buffer from the start.
@@ -566,6 +562,10 @@ static int rockchip_drm_gem_object_mmap(struct drm_gem_object *obj,
 	vma->vm_page_prot = pgprot_writecombine(vm_get_page_prot(vma->vm_flags));
 	vma->vm_page_prot = pgprot_decrypted(vma->vm_page_prot);
 
+        /* default is wc. */
+        if (rk_obj->flags & ROCKCHIP_BO_CACHABLE)
+                vma->vm_page_prot = vm_get_page_prot(vma->vm_flags);
+
 	if (rk_obj->buf_type == ROCKCHIP_GEM_BUF_TYPE_SECURE) {
 		DRM_ERROR("Disallow mmap for secure buffer\n");
 		ret = -EINVAL;
```

JeffyCN, 2024-02-04:
thanks for reporting.

 @HermanChen  please check that

HermanChen, 2024-02-04:
OK, I will let drm people check it.

JeffyCN, 2024-03-05:
Hi,
for the first issue, it's expected to use the upstream dma-heap driver(according to rk dma-heap developer(@xxm)), so we should fix the crash in the mpp @HermanChen 

and the second issue would be fixed in: kernel-6.1-2024_03_01

JeffyCN, 2024-03-05:
Okay, thanks :)

closed 2024-03-05 by JeffyCN

Joshua-Riek, 2024-03-05:
Hey @JeffyCN quick question, is kernel-6.1-2024_03_01 from the linux-6.1-stan-rkr2 release?

JeffyCN, 2024-03-05:
no, it just a random commit before the first rk3576 thing(which i was told not to upload)

Joshua-Riek, 2024-03-05:
Gotcha, thanks!