JeffyCN/mirrors #21: Libmali with Vulkan support for RK3588 (mali T620)

vassad, 2024-02-03:
Hello!

I have a question about https://github.com/tsukumijima/libmali-rockchip.git (this repo, branch libmali):
I use OrangePi5B device, trying to build on Ubuntu for -Dplatform=vulkan with error:
`The Meson build system
Version: 0.61.2
Source dir: /home/orangepi/git/libmali-rockchip
Build dir: /home/orangepi/git/libmali-rockchip/build
Build type: native build
Project name: libmali
Project version: 1.9.0
C compiler for the host machine: cc (gcc 11.4.0 "cc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
C linker for the host machine: cc ld.bfd 2.38
Host machine cpu family: aarch64
Host machine cpu: aarch64
Message: Building for aarch64|valhall|g610|g13p0|vulkan|O3
meson.build:40:2: ERROR: Problem encountered: Failed to find matched library`

As i understand, i need a prebuilt library like **libmali-valhall-g610-g13p0-vulkan-gbm.so** in lib/ folder? If so, where can i take it/build it? 
If i build with this command:
`meson build . -Dgpu=valhall -Dversion=g610 -Dsubversion=g13p0 -Dplatform=x11`
Everything is fine, EXCEPT this:
`Checking for function "vk_icdGetInstanceProcAddr" with dependency -lmali-prebuilt: NO`

T620 seems to work with vulkan cause of data/mali.json.in commit with confirmation that it works correctly.
`"Tested with an internal version of DDK for vulkan, vulkaninfo works." `

Best regards,
Vadim.

JeffyCN, 2024-02-04:
Sorry, i don't know much about that, the mali DDK is maintained by the RK GPU team internally, it seems like the vulkan library has not yet released.

If you want to know the release schedule, maybe try to mail the maintainers(the committers of the related libraries)

closed 2024-02-04 by JeffyCN

JeffyCN, 2024-02-04:
right, it has different backends, the local version i used to test was drm(kms) backend.

vassad, 2024-02-04:
> right, it has different backends, the local version i used to test was drm(kms) backend.

so right now it is possible to use mali with vulkan but only on Android?

JeffyCN, 2024-02-04:
i don't know the release schedule, the code for drm backend should be ready (at least i've tested it locally long time ago)

vassad, 2024-02-04:
got it, thx again :)