JeffyCN/mirrors #22: RGA2 Error on rk3568 When Handling Memory Larger Than 4G!

shivabohemian, 2024-03-12:
On the rk3568 platform, the RGA2 is reporting an error when handling memory larger than 4G. Is there a need to modify the `gstreamer-rockchip` or `mpp` code to address this issue? Thank you!
rga驱动版本：RGA multicore Device Driver: v1.2.25
rga API版本：rga_api version 1.10.0_[5]

[image]

JeffyCN, 2024-03-12:
Hi,

This is just a mirror repo, please report this issue to the RGA maintainer(or commiter).

closed 2024-03-12 by JeffyCN

shivabohemian, 2024-03-12:
Sorry, I only found the official `gstreamer-rockchip` repository here.
I am using this repo's latest `gstreamer-rockchip` plugin for transcoding, but the screen turns all green. Do you possibly know the reason for this? The old `gstreamer-rockchip` in the SDK doesn't have this problem. Below is roughly the video transcoding part of the gst.
`mppvideodec format=23 ! mpph265enc rc-mode=vbr width= height=500 ! h265parse`

JeffyCN, 2024-03-12:
please contact rga developers for this issue.

according to the log, you are using a device with ddr larger than 4g, and the rga might not be able to handle it.

it's a common issue, they should have a document about how to workaround it. maybe limit the total memory to 4g or limit the video memory to 4g(drm allocator or dma heap allocator). anyway, i don't know much about it, just providing what i've heard from other customers.

shivabohemian, 2024-03-12:
Thank you for your answer. By the way, the mirror repository's `gstreamer-rockchip` can be compiled and used directly, right?

JeffyCN, 2024-03-12:
right, use meson or dpkg-buildpackage

shivabohemian, 2024-03-12:
> right, use meson or dpkg-buildpackage

When I compiled the latest `gstreamer-rockchip` code, I found two problems: the first is the plugin without `mppvp8enc`, and the second is the green screen after transcoding now. 
What's going on here? My `gst` command (below) is normal in very old `gstreamer-rockchip`.
`mppvideodec format=23 ! mpph265enc rc-mode=vbr width=1280 height=720 ! h265parse`

JeffyCN, 2024-03-12:
1/ enc plugins would only be registered when mpp can init it. so if vp8enc is missing, please ask mpp maintainer.

2/ **please contact rga developers for rga issue.**

shivabohemian, 2024-03-12:
Ok, thank you very much!