JeffyCN/mirrors #24: 在ubuntu 22.04的嵌入式服务器系统上报：could not initialize canvas

kevin406972552, 2024-04-16:
在ubuntu22.04
下载源码后meson build 发现生成的.so只有8.2kb，ninja ---> ninja install 后，把安装目录的.so文件覆盖到系统/usr/lib/

JeffyCN, 2024-04-16:
rwxr-xr-x 1 root root 42M Apr 15 16:10 libmali.so.1.9.0

JeffyCN, 2024-04-16:
pls try to test it with newest glmark2

kevin406972552, 2024-04-16:
> rwxr-xr-x 1 root root 42M Apr 15 16:10 libmali.so.1.9.0
-rwxr-xr-x 1 root root 8.2K Apr 15 15:07 libEGL.so.1
-rwxr-xr-x 1 root root 8.2K Apr 15 15:07 libGLESv1_CM.so.1
-rwxr-xr-x 1 root root 8.2K Apr 15 15:07 libGLESv2.so.2 And so on ，Both are 8.2k

kevin406972552, 2024-04-16:
两个系统环境下用的同样的glmark2-es2-drm版本，都是
apt update
apt install glmark2-es2-drm 安装的。
最大的问题是在ubuntu22.04下ninja install 生成的
-rwxr-xr-x 1 root root 8.2K Apr 15 15:07 libEGL.so.1
-rwxr-xr-x 1 root root 8.2K Apr 15 15:07 libGLESv1_CM.so.1
-rwxr-xr-x 1 root root 8.2K Apr 15 15:07 libGLESv2.so.2 等等都异常的小。

我也试过只把下面的覆盖过去，也不行
lrwxrwxrwx 1 root root 17 Apr 15 15:07 libmali-hook.so -> libmali-hook.so.1
lrwxrwxrwx 1 root root 21 Apr 15 15:07 libmali-hook.so.1 -> libmali-hook.so.1.9.0
-rwxr-xr-x 1 root root 23K Apr 15 15:07 libmali-hook.so.1.9.0
lrwxrwxrwx 1 root root 12 Apr 15 15:07 libmali.so -> libmali.so.1
lrwxrwxrwx 1 root root 16 Apr 15 15:07 libmali.so.1 -> libmali.so.1.9.0
-rwxr-xr-x 1 root root 42M Apr 15 15:07 libmali.so.1.9.0

JeffyCN, 2024-04-16:
they are expected to be small.

pls use the same glmark2 to test.

kevin406972552, 2024-04-16:
在ubuntu23.10的文件系统情况下，刚刚还做了Qt的交叉编译：
1：在没更换编译出来的库时，config正常通过，并支持opengl
2：在更换编译出来的库后却报egl，eglfs，opengl es2 依赖库的错误

JeffyCN, 2024-04-20:
For Debian-based OSs, this libmali uses vendor packaging, which would conflict with the unified libmali library.
So there would be some dummy(smallest) libraries to wrap the libmali.
This requires third-party packages to get compiling flags through pkg-config instead of being hard-coded in their build scripts.

The suggested way to use this libmali on Debian-based OS would be:
1/ Remove it when compiling other packages.
2/ Install it only when you are about to use it.

closed 2024-04-20 by JeffyCN