JeffyCN/mirrors #30: Hardware Accelerated Deinterlacing Support

jesus2k21, 2024-09-09:
Apologies if this isn't the right place to ask this question, but I have been looking for the capability of deinterlacing video with GStreamer on rk3588 and have not found much on the topic. Are there such capabilities for hardware accelerated deinterlacing using GStreamer?

The [MPP development guide](https://github.com/rockchip-linux/mpp/blob/develop/doc/Rockchip_Developer_Guide_MPP_EN.md) seems to indicate that the platform is capable of deinterlacing but I have not found any way to do so using GStreamer.

JeffyCN, 2024-09-09:
normally it should be done in the mpp, if it doesn't work, you can report to the mpp's maintainer.

jesus2k21, 2024-09-09:
Understood, I'll reach out to the mpp maintainers. Thanks!

closed 2024-09-09 by jesus2k21