JeffyCN/mirrors #32: mpph264enc may crash

niedavy, 2024-10-31:
OS: Armbian (24.2.1) aarch64 
Host: Orange Pi 5 
Kernel: 5.10.160-legacy-rk35xx 

mpp version:  [GitHub - rockchip-linux/mpp: Media Process Platform (MPP) module](https://github.com/rockchip-linux/mpp/tree/develop)

[mpph264enc](https://github.com/JeffyCN/mirrors/tree/gstreamer-rockchip/gst/rockchipmpp) may crash when seeking. The reason is that gst_video_encoder_get_oldest_frame returns a null pointer.

sample code: 
```c++
while (true) {

	std::string pipeline_description = fmt::format(
	    "videotestsrc ! video/x-raw,format=NV12 ! "
	    "mpph264enc bps-max=500000 max-reenc=1 gop=50  ! h264parse ! video/x-h264, "
	    "alignment=(string)au ! fakesink name=fakesink sync=false "
	    "signal-handoffs=true sync=true");
	...;

	pipeline = gst_parse_launch(pipeline_description.c_str(), nullptr);

	... ;

	gst_element_set_state(pipeline, GST_STATE_PLAYING);
	gst_element_get_state(pipeline, nullptr, nullptr, GST_CLOCK_TIME_NONE);

	if (FALSE == gst_element_seek(pipeline, 1.0, GST_FORMAT_TIME,
	                              GstSeekFlags(GST_SEEK_FLAG_FLUSH | GST_SEEK_FLAG_KEY_UNIT), GST_SEEK_TYPE_SET,
	                              10 * GST_NSECOND, GST_SEEK_TYPE_NONE, GST_CLOCK_TIME_NONE)) {
	...;
	}

	...;
	if (pipeline != nullptr) {
		gst_element_set_state(pipeline, GST_STATE_PAUSED);
		gst_element_get_state(pipeline, nullptr, nullptr, GST_SECOND);
		gst_element_set_state(pipeline, GST_STATE_READY);
		gst_element_get_state(pipeline, nullptr, nullptr, GST_SECOND);
		gst_element_set_state(pipeline, GST_STATE_NULL);
		gst_element_get_state(pipeline, nullptr, nullptr, GST_SECOND);
		gst_object_unref(pipeline);
		pipeline = nullptr;
	}
}
```

JeffyCN, 2024-10-31:
thanx for reporting. would this help?
```
gst/rockchipmpp/gstmppenc.c
@@ -1023,7 +1023,7 @@ gst_mpp_enc_poll_packet_locked (GstVideoEncoder * encoder)
   /* This encoded frame must be the oldest one */
   frame = gst_video_encoder_get_oldest_frame (encoder);

-  if (self->flushing && !self->draining)
+  if (!frame || self->flushing && !self->draining)
     goto drop;
```

niedavy, 2024-11-01:
not working. After using this method, encode_get_packet always returns null.

JeffyCN, 2024-11-01:
could you provide a prebuilt test program to reproduce it?

niedavy, 2024-11-01:
https://github.com/niedavy/mppenc_test is test code.

JeffyCN, 2024-11-04:
should be fixed now

niedavy, 2024-11-05:
in my environment, this solution is not working. Do I need to add some environment variables?

GStreamer: 1.20.3
OS: Armbian (24.2.1) aarch64
Host: Orange Pi 5
Kernel: 5.10.160-legacy-rk35xx
mpp: [GitHub - rockchip-linux/mpp: Media Process Platform (MPP) module](https://github.com/rockchip-linux/mpp/tree/develop)
[image]

JeffyCN, 2024-11-05:
no, it should be fixed by https://github.com/JeffyCN/mirrors/commit/17e8bcef39a9a9d62d87c83763de0e2428fc9406

when seeking, the plugin will flush the pending frames, which was:
1/ reset the mpp
2/ discard framework's frame list

but it looks like the mpp reset didn't clear it's internal frame list, so it will output unexpected frames.

that patch is to wait for all pending frames before flushing.

niedavy, 2024-11-05:
When gst_mpp_enc_reset(encoder, FALSE, FALSE) is called within gst_mpp_enc_flush, the 'drain' parameter is set to false. Consequently, in gst_mpp_enc_stop_task,  self->pending_frames is directly reset to 0, leading gst_mpp_enc_loop to immediately proceed to 'goto out'. 

I try to set the 'drain' parameter to TRUE in gst_mpp_enc_flush. For now, test programs run well.
```
 gst_mpp_enc_flush (GstVideoEncoder * encoder)
 {
   GST_DEBUG_OBJECT (encoder, "flushing");
-  gst_mpp_enc_reset (encoder, FALSE, FALSE);
+  gst_mpp_enc_reset (encoder, TRUE, FALSE);
   return TRUE;
 }
```

JeffyCN, 2024-11-05:
hmm, thanx for debuging it, that is the same as my original test code:)

i've re-uploaded the patch.

niedavy, 2024-11-05:
thanx, all done

closed 2024-11-05 by niedavy