JeffyCN/mirrors #34: help with libmali on Alwwinner H700 (Mali G31)

loki666, 2024-11-20:
I understand this project is about Rockchip, so please disregard this issue if not appropriate.

I'm currently trying to get the G31 running on kernel 6.12
kernel driver "seems" to run fine, it correctly identifies the GPU and probe it ok.

but applications fails in some various ways...

currently my best setup is using your libmali-bifrost-g31-g2p0-gbm.so with gl2mark
the first two tests pass ok (I can see the rotating horse on the display), but the nexts batch of tests (texture-nearest,...) all fails.
nothing is displayed on screen and the kernel driver reports failing jobs with the ERROR_INVALID_DATA
shading tests seems also ok

I understand that it's the role of the userspace lib to prepare these jobs buffer and the kernel driver merely launch them on GPU, do you have an idea to help me with this?

Thanks in advance.

loki666, 2024-11-20:
here is a log
```
Debug: Reading PNG file /usr/share/glmark2/textures/crate-base.png
Debug:     Height: 512 Width: 512 Bpp: 3
[  169.846282] mali 1800000.gpu: error detected from slot 0, job status 0x00000058 (DATA_INVALID_FAULT)
[  169.856598] mali 1800000.gpu: t6xx: GPU fault 0x58 from job slot 0
Debug: Loading vertex shader from file None:
Debug: #if defined(GL_ES)
Debug: #define HIGHP_OR_DEFAULT highp
Debug: #else
Debug: #define HIGHP_OR_DEFAULT
Debug: #endif
Debug: #if defined(GL_ES)
Debug: #define MEDIUMP_OR_DEFAULT mediump
Debug: #else
Debug: #define MEDIUMP_OR_DEFAULT
Debug: #endif
Debug: const vec4 MaterialDiffuse = vec4(1.000000, 1.000000, 1.000000, 1.000000);
Debug: const vec4 LightSourcePosition = vec4(20.000000, 20.000000, 10.000000, 1.000000);
Debug: attribute vec3 position;
Debug: attribute vec3 normal;
Debug: attribute vec2 texcoord;
Debug: 
Debug: uniform mat4 ModelViewProjectionMatrix;
Debug: uniform mat4 NormalMatrix;
Debug: 
Debug: varying vec4 Color;
Debug: varying vec2 TextureCoord;
Debug: 
Debug: void main(void)
Debug: {
Debug:     // Transform the normal to eye coordinates
Debug:     vec3 N = normalize(vec3(NormalMatrix * vec4(normal, 1.0)));
Debug: 
Debug:     // The LightSourcePosition is actually its direction for directional light
Debug:     vec3 L = normalize(LightSourcePosition.xyz);
Debug: 
Debug:     // Multiply the diffuse value by the vertex color (which is fixed in this case)
Debug:     // to get the actual color that we will use to draw this vertex with
Debug:     float diffuse = max(dot(N, L), 0.0);
Debug:     Color = vec4(diffuse * MaterialDiffuse.rgb, MaterialDiffuse.a);
Debug: 
Debug:     // Set the texture coordinates as a varying
Debug:     TextureCoord = texcoord;
Debug: 
Debug:     // Transform the position to clip coordinates
Debug:     gl_Position = ModelViewProjectionMatrix * vec4(position, 1.0);
Debug: }
Debug: Loading fragment shader from file None:
Debug: #if defined(GL_ES) && defined(GL_FRAGMENT_PRECISION_HIGH)
Debug: #define HIGHP_OR_DEFAULT highp
Debug: #else
Debug: #define HIGHP_OR_DEFAULT
Debug: #endif
Debug: #if defined(GL_ES)
Debug: #define MEDIUMP_OR_DEFAULT mediump
Debug: #else
Debug: #define MEDIUMP_OR_DEFAULT
Debug: #endif
Debug: #ifdef GL_ES
Debug: precision mediump float;
Debug: #endif
Debug: uniform sampler2D MaterialTexture0;
Debug: 
Debug: varying vec4 Color;
Debug: varying vec2 TextureCoord;
Debug: 
Debug: void main(void)
Debug: {
Debug:     vec4 texel = texture2D(MaterialTexture0, TextureCoord);
Debug:     gl_FragColor = texel * Color;
Debug: }
Debug: 
Debug: Loading model from 3ds file '/usr/share/glmark2/models/cube.3ds'
Debug:     Object name: Cube Vertex count: 20 Face count: 12
Info: [texture] texture-filter=linear:^C FPS: 1428 FrameTime: 0.700 ms
```

it's seems to hapens evertime a texture is loaded

JeffyCN, 2024-11-20:
don't know much about that, but the there are many versions of libmali and mali driver, you should use the matched ones.

maybe you can try to use the open source mali driver(panfrost) and mesa.

closed 2024-11-20 by JeffyCN