JeffyCN/mirrors #37: The latest libmali for G610 fails to expose `vkGetBufferMemoryRequirements2`

ginkage, 2024-12-07:
One way to reproduce is to run Dolphin, which fails in  in `VmaAllocator_T::GetBufferMemoryRequirements` from the `VulkanMemoryAllocator` external library.
Another way is to run AetherSX2, which fails to create device context and spits out warnings:
```
Vulkan: Failed to load required device function vkGetBufferMemoryRequirements2
Vulkan: Failed to load required device function vkGetImageMemoryRequirements2
Vulkan: Failed to load required device function vkBindBufferMemory2
```
Does `vk_icdGetInstanceProcAddr` actually work properly for these extensions?

JeffyCN, 2024-12-08:
sorry, i don't know much about it, maybe you can mail the g24 g610 libmali's committor and provide a test demo or rootfs image to reproduce this issue.

ginkage, 2024-12-08:
@JeffyCN Um, but you are the one who committed g24 here?..
Or do you mean Zhixiong Lin (zhixiong.lin@rock-chips.com)?

ginkage, 2024-12-08:
@linzhixiong1996 Could you please have a look at this?

JeffyCN, 2024-12-08:
> @JeffyCN Um, but you are the one who committed g24 here?.. Or do you mean Zhixiong Lin ([zhixiong.lin@rock-chips.com](mailto:zhixiong.lin@rock-chips.com))?

right, the original committor(author, of the fisrt signed-of-by). i can only handle issue about the mali-hook part and scripts.

and by the way, most of the rockchip guys not use github, you can send a mail as well to make sure that they can see the issue.

StonedEdge, 2024-12-09:
> right, the original committor(author, of the fisrt signed-of-by). i can only handle issue about the mali-hook part and scripts.
>
> and by the way, most of the rockchip guys not use github, you can [...]

We've sent an email to zhixiong.lin@rock-chips.com & chenzhen@rock-chips.com (hopefully they are the right people). Please forward to any other relevant Rockchip employees so we could get this bug fixed.

HeyMeco, 2024-12-09:
I was able to reproduce this issue, looking forward to a solution

ginkage, 2024-12-10:
I was able to work around this problem by modifying the hook...

In short, looks like the method pointers for the symbols above are *not* returned from `vkGetDeviceProcAddr`, yet the address is returned successfully from `vk_icdGetInstanceProcAddr`. Hence my workaround is to fall back to `vk_icdGetInstanceProcAddr` when the driver is about to return NULL.

This approach fixes both AetherSX2 and Dolphin for me:

```
diff --git a/hook/hook.c b/hook/hook.c
index 9f42e8a..b9a4d2a 100644
--- a/hook/hook.c
+++ b/hook/hook.c
@@ -42,6 +42,10 @@
 #include <X11/Xlibint.h>
 #endif
 
+#ifdef HAS_VULKAN
+#include <vulkan/vulkan.h>
+#endif
+
 #ifndef ARRAY_SIZE
 #define ARRAY_SIZE(x) (sizeof(x)/sizeof(x[0]))
 #endif
@@ -112,6 +116,10 @@ static EGLBoolean (* _eglDestroySurface)(EGLDisplay dpy, EGLSurface surface) = N
 static PFNEGLMAKECURRENTPROC _eglMakeCurrent = NULL;
 #endif
 
+#ifdef HAS_VULKAN
+static PFN_vkVoidFunction (* _vk_icdGetInstanceProcAddr)(VkInstance instance, const char* pName) = NULL;
+#endif
+
 #define MALI_SYMBOL(func) { #func, (void **)(&_ ## func), }
 static struct {
    const char *func;
@@ -166,6 +174,9 @@ static struct {
    MALI_SYMBOL(eglDestroySurface),
    MALI_SYMBOL(eglMakeCurrent),
 #endif
+#ifdef HAS_VULKAN
+   MALI_SYMBOL(vk_icdGetInstanceProcAddr),
+#endif
 };
 
 __attribute__((constructor)) static void
@@ -859,3 +870,30 @@ eglGetProcAddress(const char *procname)
 }
 
 #endif // HAS_EGL
+
+#ifdef HAS_VULKAN
+typedef void * (* PFNvkGetDeviceProcAddr)(struct VkDevice_T *, const char *);
+static PFNvkGetDeviceProcAddr _vkGetDeviceProcAddr = NULL;
+static VkInstance _vk_instance;
+
+void* myGetDeviceProcAddr(VkDevice device, const char* pName) {
+   void *addr = _vkGetDeviceProcAddr(device, pName);
+   fprintf(stderr, "*** GinHook %p %s: %p\n", device, pName, addr);
+   if (addr == NULL) {
+      addr = _vk_icdGetInstanceProcAddr(_vk_instance, pName);
+      fprintf(stderr, "!!! GinHook %p %s: %p\n", device, pName, addr);
+   }
+   return addr;
+}
+
+VKAPI_ATTR PFN_vkVoidFunction VKAPI_CALL vk_icdGetInstanceProcAddr(VkInstance instance, const char* pName) {
+   void *addr = _vk_icdGetInstanceProcAddr(instance, pName);
+   fprintf(stderr, "GinHook %p %s: %p\n", instance, pName, addr);
+   if (!strcmp(pName, "vkGetDeviceProcAddr")) {
+      _vk_instance = instance;
+      _vkGetDeviceProcAddr = addr;
+      return (PFN_vkVoidFunction)myGetDeviceProcAddr;
+   }
+   return addr;
+}
+#endif
diff --git a/hook/meson.build b/hook/meson.build
index 9243ff0..2760207 100644
--- a/hook/meson.build
+++ b/hook/meson.build
@@ -59,6 +59,10 @@ if cc.has_function('eglCreateContext', dependencies : mali)
   endif
 endif
 
+if cc.has_function('vk_icdGetInstanceProcAddr', dependencies : mali)
+  libhook_cflags += '-DHAS_VULKAN'
+endif
+
 libhook = shared_library(
   'mali-hook',
   'hook.c',
```

JeffyCN, 2024-12-19:
please mail the rockchip ddk maintainers to fix this in the ddk.

ginkage, 2024-12-19:
We did... They didn't reply.

JeffyCN, 2024-12-19:
maybe add more maintainers:

Zhixiong Lin [zhixiong.lin@rock-chips.com](mailto:zhixiong.lin@rock-chips.com)
Jacob Chen [cc@rock-chips.com](mailto:cc@rock-chips.com)
Zhen Chen [chenzhen@rock-chips.com](mailto:chenzhen@rock-chips.com)

putin.li@rock-chips.com
randall.zhuo@rock-chips.com

JeffyCN, 2024-12-20:
@Caesar-github

JeffyCN, 2025-01-08:
should be fixed in the newest version.

the api version in the vulkan icd json should be increased.

closed 2025-01-08 by JeffyCN