JeffyCN/mirrors #39: libmali g24p0-4 incorrect colors (gamma) on Mutter (GNOME Wayland) when using vulkan renderer

StonedEdge, 2024-12-18:
We've been testing the latest g24p0 blob drivers with Vulkan and I'm finding that the colors are not rendering correctly with Mutter, but work fine with Weston. 

Is this something that Rockchip are able to reproduce with the Mutter compositor? This happens in multiple apps (AetherSX2, Lime3DS and Chromium when trying to run with the Vulkan renderer). My guess it is something incorrect with either the swapchain configuration, or some sort of mismatch between rendering output of Vulkan and expected colorspace for Mutter. 

[image]

Vulkaninfo output:
https://pastebin.com/ZvCy3YkH

Current WSI layer: 
https://drive.google.com/file/d/1sor29bVrSe_wCUE4ptDZYwoqyk6fDmN_/view?usp=sharing

JeffyCN, 2024-12-20:
@Caesar-github

StonedEdge, 2025-01-08:
FWIW, this issue seems to only happen on a DSI display. With each screen mirroring one another, the issue becomes more obvious. The external monitor renders correctly but the DSI screen does not... I wonder if this is a quirk of the Rockchip DSI driver itself. 

https://github.com/user-attachments/assets/c146be0e-6e19-46fc-8e82-a5b13f2be423

JeffyCN, 2025-01-10:
don't know much about that, but if it only happen on DSI, maybe not related to GPU.

maybe try to use kmsgrab to dump the display buffer:
https://filebin.net/53l5wb21wua42cg4/kmsgrab

if the buffer is correct, should be something wrong with the DSI, then you can contact the RK DSI maintainers(check the kernel driver's commit log)

ginkage, 2025-01-11:
Haven't tried kmsgrab yet, but there is something I want to point out here.
The actual issue is not about gamma as such, the issue is that the swapchain buffers are rendered as translucent instead of being opaque. You can see it by running e.g. vkcube: even under Weston, it will be displayed in a semi-transparent box.
The difference between HDMI and DSI there is that HDMI has the whole screen buffer cleared every frame, whereas DSI keeps layering those translucent frames on top of each other, causing eventual color burnout.
So, the question is, how can that translucency be fixed? It could be a WSI issue, of course: we're using the [standard](https://gitlab.freedesktop.org/mesa/vulkan-wsi-layer/-/tree/main?ref_type=heads) one, and ARM doesn't post the modifications they did to it (and I know for a fact that they did modify it). I wonder if there's a Rockchip internal repo with the WSI for G610, and if that can be published.

JeffyCN, 2025-01-11:
Arm doesn't allow to publish code.

Maybe you can ask jacob about it:
Jacob Chen [cc@rock-chips.com](mailto:cc@rock-chips.com)

ginkage, 2025-01-11:
Why though? In theory, they should be doing it: https://developer.arm.com/downloads/-/mali-drivers/vulkan-wsi-layer
In practice, that page hosts some ancient sources without actual changes.

JeffyCN, 2025-01-13:
no idea about that, please mail the maintainers for detail.

ginkage, 2025-02-06:
Anyway, we've found a workaround for this. It still seems like an incorrect behaviour, but oh well, I guess the driver itself is unlikely to be fixed.

JeffyCN, 2025-02-06:
maybe you can provide a rootfs image to reproduce it to jacob, he is the maintainer of rockchip's libmali g24p0.

closed 2025-04-21 by JeffyCN