JeffyCN/mirrors #44: [libmali] G610 New version does not load firmware

St-Viacheslavik, 2025-02-04:
Hi

For RK3588 I used to build libmali package from commit `92183c8482e6173fa510f228e62b1c73c99be87d`
https://github.com/JeffyCN/mirrors/tree/92183c8482e6173fa510f228e62b1c73c99be87d
When starting X server kernel driver loaded firmware

[image]
Graphics work with GL4ES

[image]

Now I updated to the latest commit and now kernel does not load mali_csffw.bin and all graphics work only on ARM CPU (using llvmpipe)

[image]

What could go wrong?
Can you tell me in what order the packages need to be collected for the graphics to work correctly via Mali? I am currently building everything for Debian 11 using the following repository mirrors
## XServer:
Debian 11: https://github.com/JeffyCN/xorg-xserver/tree/rockchip/debian/1.20.11
## DRM Cursor:
https://githugitb.com/JeffyCN/drm-cursor
## MPV:
https://github.com/JeffyCN/mpv/tree/debian/0.29.1-1
## Libv4l-rkmpp:
https://github.com/JeffyCN/libv4l-rkmpp
To build Libv4l-rkmpp you need to clone `v4l` and patch it first.
Sources:
https://salsa.debian.org/debian/libv4l.git
**Tag `debian/1.22.1-4`**
Patch:
https://github.com/JeffyCN/meta-rockchip/blob/release-1.3.0_20200915/recipes-multimedia/v4l2apps/v4l-utils/0001-libv4l2-Support-mmap-to-libv4l-plugin.patch
## gstreamer-rockchip:
https://github.com/JeffyCN/mirrors/tree/gstreamer-rockchip
## RGA:
https://github.com/JeffyCN/mirrors/tree/linux-rga-multi
## MALI:
https://github.com/JeffyCN/mirrors/tree/libmali
## MPP:
https://github.com/rockchip-linux/mpp
## GL4ES:
https://github.com/ptitSeb/gl4es

In what order should these components be assembled?
And what are the differences between Mali versions g6p0 g13p0 g24p0, how do I determine which version to use for RK3588?

JeffyCN, 2025-02-04:
the libmali works with kernel mali driver, don't know much about the details, but maybe try to update your kernel.

you can also try to mail the commit author to get the exact kernel commit :)

St-Viacheslavik, 2025-02-06:
> For the latest libmali in this repo, you need a kernel with DDK g25, namely kernel 6.1-rkr5. That kernel has the firmware built-in (so you don't need that file anymore), and also has a couple commits that add `system-uncached` DMA heap which is required by libmali g24-6.

Yes, I took a new kernel from Armbian, after that Mali initializes successfully
The question is a bit abstract now. For testing I used glxheads and glxgears, for Mali to work it was necessary to use the library /usr/lib/gl4es/libGL.so.1 from PitSeb

For testing I have standard glxgears and glmark2-es2
Both applications display different openGL Render. For glxheads `GL4ES wrapper`. For glmark `Mali-G610`
For `glxheads` `glxgears`
```bash
glxheads
Name: :0
  Display:     0x55972288a0
  Window:      0x2400002
  Context:     0x5596c20280
  GL_VERSION:  2.1 gl4es wrapper 1.1.7
  GL_VENDOR:   ptitSeb
  GL_RENDERER: GL4ES wrapper
vblank_mode=0 glxgears
LIBGL: Initialising gl4es
LIBGL: v1.1.7 built on Jul 30 2021 06:53:15
LIBGL: Using GLES 2.0 backend
LIBGL: loaded: libGLESv2.so
LIBGL: loaded: libEGL.so
LIBGL: Using GLES 2.0 backend
arm_release_ver: g24p0-00eac0, rk_so_ver: 6
LIBGL: Hardware Full NPOT detected and used
LIBGL: Extension GL_EXT_blend_minmax  detected and used
LIBGL: FBO are in core, and so used
LIBGL: PointSprite are in core, and so used
LIBGL: CubeMap are in core, and so used
LIBGL: BlendColor is in core, and so used
LIBGL: Blend Subtract is in core, and so used
LIBGL: Blend Function and Equation Separation is in core, and so used
LIBGL: Texture Mirrored Repeat is in core, and so used
LIBGL: Extension GL_OES_mapbuffer  detected
LIBGL: Extension GL_OES_element_index_uint  detected and used
LIBGL: Extension GL_OES_packed_depth_stencil  detected and used
LIBGL: Extension GL_OES_depth24  detected and used
LIBGL: Extension GL_OES_rgb8_rgba8  detected and used
LIBGL: Extension GL_EXT_texture_format_BGRA8888  detected and used
LIBGL: Extension GL_OES_depth_texture  detected and used
LIBGL: Extension GL_OES_texture_stencil8  detected and used
LIBGL: Extension GL_EXT_texture_rg  detected and used
LIBGL: Extension GL_EXT_color_buffer_float  detected and used
LIBGL: Extension GL_EXT_color_buffer_half_float  detected and used
LIBGL: high precision float in fragment shader available and used
LIBGL: Max vertex attrib: 32
LIBGL: Extension GL_OES_standard_derivatives  detected and used
LIBGL: Extension GL_ARM_shader_framebuffer_fetch detected and used
LIBGL: Extension GL_OES_get_program_binary  detected and used
LIBGL: Number of supported Program Binary Format: 1
LIBGL: Max texture size: 16383
LIBGL: Max Varying Vector: 31
LIBGL: Texture Units: 16/16 (hardware: 64), Max lights: 8, Max planes: 6
LIBGL: Extension GL_EXT_texture_filter_anisotropic  detected and used
LIBGL: Max Anisotropic filtering: 16
LIBGL: Max Color Attachments: 1 / Draw buffers: 1
LIBGL: Hardware vendor is ARM
LIBGL: sRGB surface supported
LIBGL: EGLImage from Pixmap supported
LIBGL: EGLImage to Texture2D supported
LIBGL: EGLImage to RenderBuffer supported
LIBGL: Targeting OpenGL 2.1
LIBGL: Not trying to batch small subsequent glDrawXXXX
LIBGL: Trying to use VBO
LIBGL: glXMakeCurrent FBO workaround enabled
LIBGL: FBO workaround for using binded texture enabled
LIBGL: Force texture for Attachment color0 on FBO
LIBGL: Hack to trigger a SwapBuffers when a Full Framebuffer Blit on default FBO is done
LIBGL: glX Will try to recycle EGL Surface
LIBGL: Current folder is:/home/rc_module
LIBGL: Loaded a PSA with 1 Precompiled Programs
arm_release_ver: g24p0-00eac0, rk_so_ver: 6
Running synchronized to the vertical refresh.  The framerate should be
approximately the same as the monitor refresh rate.
150 frames in 5.0 seconds = 29.923 FPS
```

glmark2-es2:
```bash
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      ARM
    GL_RENDERER:    Mali-G610
    GL_VERSION:     OpenGL ES 3.2 v1.g24p0-00eac0.6b39a7384565dd0742f84d5ab1cdf9ef
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   800x600 windowed
=======================================================
```

Is it possible to make glxgears work via Mali-G610?
And how to disable vertical sync in gl4es to get maximum FPS?

ginkage, 2025-02-06:
It already works via libmali (you can tell by `arm_release_ver: g24p0-00eac0, rk_so_ver: 6`), so what exactly is your question here?

St-Viacheslavik, 2025-02-06:
> It already works via libmali (you can tell by `arm_release_ver: g24p0-00eac0, rk_so_ver: 6`), so what exactly is your question here?

I get different FPS values ​​when testing through glxgears and glmark2.
`glxgears` *30FPS* with VSYNC option. (I don't know how to disable this option. `vblank_mode=0` not works)
`glmark2-es2` up to *450FPS*

closed 2025-02-07 by St-Viacheslavik