JeffyCN/mirrors #46: gstreamer h264 编码发生断错误

cppcoffee, 2025-02-26:
在 RK3588 使用 GStreamer 做 h264 编码，出现了断错误:

```shell
# gst-launch-1.0 -e v4l2src device=/dev/video0 io-mode=4 ! videoconvert ! video/x-raw,format=NV12,width=1920,height=1080 ! mpph264enc ! filesink location=output2.mp4
Setting pipeline to PAUSED ...
Using mplane plugin for capture
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Caught SIGSEGV
#0  0x0000007fa05ebe38 in poll () at /lib/aarch64-linux-gnu/libc.so.6
#1  0x0000007fa07eae58 in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#2  0x0000007fa0795024 in g_main_loop_run ()
#3  0x000000558b9641a0 in  ()
#4  0x0000007fa05373fc in  () at /lib/aarch64-linux-gnu/libc.so.6
#5  0x0000007fa05374cc in __libc_start_main ()
#6  0x000000558b9642f0 in _start ()
Segmentation fault (core dumped)
```

cppcoffee, 2025-02-28:
```root@orangepi5plus:~# gst-launch-1.0 v4l2src device=/dev/video0 !     video/x-raw,width=1920,height=1080 !     videoconvert !     mppjpegenc !     multipartmux !     tcpserversink host=0.0.0.0 port=5000Setting pipeline to PAUSED ...
Using mplane plugin for capture
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Caught SIGSEGV
#0  0x0000007f8fe9be38 in poll () at /lib/aarch64-linux-gnu/libc.so.6
#1  0x0000007f9009ae58 in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#2  0x0000007f90045024 in g_main_loop_run ()
#3  0x000000556f9141a0 in  ()
#4  0x0000007f8fde73fc in  () at /lib/aarch64-linux-gnu/libc.so.6
#5  0x0000007f8fde74cc in __libc_start_main ()
#6  0x000000556f9142f0 in _start ()
Spinning.  Please run 'gdb gst-launch-1.0 25924' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:42.185251880
Setting pipeline to NULL ...

^C
```

cppcoffee, 2025-02-28:
5.10 版本的内核就正常运行

JeffyCN, 2025-04-21:
Could not reproduce it, and your kernel(6.1.43) is too old.

closed 2025-04-21 by JeffyCN