JeffyCN/mirrors #49: libmali: upgrade G610 userspace driver to g25p0

RadxaStephen, 2025-03-13:
Hi Jeffy

I find that the newer rk kernel upgrade bifrost DDK to g25p0-00eac0.

```
commit d1b62d2e45b49ef7792d43a660d048d8618dee46
Author: Zhen Chen <chenzhen@rock-chips.com>
Date:   Mon Aug 12 15:30:09 2024 +0800

    MALI: rockchip: upgrade bifrost DDK to g25p0-00eac0, from g24p0-00eac0
    
    mali_csffw.bin from Valhall DDK g25(r50) is included.
    
    Change-Id: Ic454428c384456a14b29d9651f537eb59c11284d
    Signed-off-by: Zhen Chen <chenzhen@rock-chips.com>
```

Can you update libmali repository to upgrade G610 userspace driver to g25p0?

JeffyCN, 2025-03-14:
i don't see any update in the libmali, the newest version is still g24p0.

RadxaStephen, 2025-03-14:
Okay. Thanks.

closed 2025-03-14 by RadxaStephen