JeffyCN/mirrors #50: gst-launch cause memory leaking

nascs, 2025-03-20:
内核版本： Linux rockchip-rk3588-rock-5b 6.1.84-rockchip-standard #1 SMP Thu Mar 13 02:31:31 UTC 2025 aarch64 GNU/Linux

复现步骤：

在 Radxa ROCK 5B yocto 上执行以下命令：

```bash
gst-launch-1.0 videotestsrc ! video/x-raw,width=3840,height=2160,framerate=15/1,format=UYVY ! tee name=t ! queue max-size-time=1000000000 ! mpph264enc ! fakesink t. ! queue max-size-time=1000000000 ! videoconvert ! autovideosink
```

之后内核崩溃，再执行其他命令会卡死或者出现段错误
top 下 Mem used 飙升： 402068K -> 468128K

串口报错：

https://gist.github.com/nascs/105549cf3a0d1be63bef76bf22d650cc

JeffyCN, 2025-03-20:
works well on my side.

nascs, 2025-03-20:
> works well on my side.

Follow the steps above, is it okay on your end?

JeffyCN, 2025-03-20:
```
root@rk3588-evb:/# gst-launch-1.0 videotestsrc ! video/x-raw,width=3840,height=2160,framerate=15/1,format=UYVY ! tee name=t ! queue max-size-time=1000000000 ! mpph264enc ! fakesink t. ! queue max-size-time=1000000000 ! videoconvert ! autovideosink
设置暂停管道 ...
管道正在 PREROLLING ...
重新分配延迟时间...9.
管道被 PREROLLED ...
Prerolled, waiting for async message to finish...
设置播放管道 ...
重新分配延迟时间...
New clock: GstSystemClock
^Chandling interrupt.
中断: 中止管道 ...
Execution ended after 0:39:02.543571617
设置 NULL 管道 ...
释放管道资源 ...
```

the output of 'free' command is stable.

nascs, 2025-03-20:
Can how tell me gst-launch version on the evb board?

JeffyCN, 2025-03-20:
gstreamer-rockchip# git log --oneline
27ba1c7 mppdec: Clear pending frames after reset

root@rk3588-evb:/# gst-launch-1.0 --version
gst-launch-1.0 version 1.22.12
GStreamer 1.22.12

nascs, 2025-03-20:
then can you tell me the kernel version and branch you used so that I can align the environment.

JeffyCN, 2025-03-20:
kernel is internal 6.1 branch:
9e2ef257d613 arm64: dts: rockchip: rk3576-evb2: Remove vcc3v3_lcd_n
35f91837231f mmc: dw_mmc: Explicitly include linux/pinctrl/consumer.h for pinctrl_select_state
47102b506abc regulator: rk806: Explicitly include linux/pinctrl/consumer.h
7166d5f4bb74 mfd: rk806: Explicitly include linux/pinctrl/consumer.h
4f984d88900c ASoC: aw87xxx: Explicitly include vmalloc.h

JeffyCN, 2025-03-20:
external/mpp# git log --oneline
6bcfb2c0 fix[av1]: parameter is 16 bits
a4326e2c feat[kmpp_obj]: Add more kmpp_obj functions
729e57a6 fix[base]: Fix strncpy compile warning
1b0aef1d fix[hal_h265e_vepu580]: Fix overflow status check
2f2e5f5c fix[kmpp]: Fix channel dup issue

closed 2025-04-21 by JeffyCN