JeffyCN/mirrors #52: GStreamer 插件 videoconvet、videosacle 通过环境变量设置 rga 加速无效

ThinkWD, 2025-04-21:
通过 gstreamer-rockchip 分支成功编译安装了 mpp 相关编解码插件，直接在编解码插件中设置转换格式是可以正常使用 rga 加速的。

但 videoconvet、videosacle 设置环境变量没有效果，检查 debug 消息也不存在任何 rga 相关字样。

如何为这些插件启用 rga 加速，是需要编译安装这些插件的特殊版本吗？

JeffyCN, 2025-04-21:
RGA has many limitations, it has been disabled by default in those GStreamer plugins. If needed, you may refer to the relevant patches in this directory:
https://github.com/JeffyCN/meta-rockchip/tree/master/recipes-multimedia/gstreamer

ThinkWD, 2025-04-21:
I'm aware of these patches, but my system is currently using the pre-installed gstreamer 1.16.3 version. Due to various constraints, I can't easily upgrade the version. Since these patches target at least 1.20.x versions, could they be backported to work with 1.16.3?

If they're compatible with 1.16.3, would the correct procedure be:
1. Clone gst-plugins-base-1.16.3 via git
2. Apply the patches
3. Compile the modified version
4. Replace the existing libgstvideoconvert.so and libgstvideoscale.so libraries in the system?

JeffyCN, 2025-04-25:
Right.

closed 2025-05-27 by JeffyCN